rodgomesc/avell-unofficial-control-center #3: Suggestion: repository renaming

luizfschrickte, 2019-05-24:
Hi Rodrigo,

I think there are many other Avell Linux users out there and it is really hard to find this repository due to its name. What do you think about renaming it to avell-unofficial-control-center or something like that? Just a suggestion, the son is yours! :)

Regards,
Luiz Fernando

rodgomesc, 2019-05-24:
when I created the repository it had exactly that name, but I was afraid to cause any  problem, put the name of a private  company in title of the project

luizfschrickte, 2019-05-24:
Got it!

When I bought the computer I've asked Avell's agents about Linux support
and they denied it - so I told them I will maybe start some kind of blog
aiming installation issues and other stuff (by that time I had hard time
installing Ubuntu due to NVIDIA video card issues). They were happy and
asked me to inform them about it so they can inform other customers - so I
think they would like your initiative (as it will probably help them sell
more computers).

I think it is safe to put their name on it if you keep the "unofficial" -
and if they don't like they will probably just ask to change it :)

We can even talk to them so maybe they offer the repository as an option
for support on their website - this is quite common among other brands (...
there is this site, it is unofficial but may help Linux users using our
devices...).

But this is up to you, just trying to be useful!

Regards

Luiz Fernando Schrickte

rodgomesc, 2019-05-24:
You can do as an example the ckb-next project, which only controls the corsair so why does not it load the name in the title? this led me to one such decision, but I may be wrong I needed to separate time to analyze

> when I created the repository it had exactly that name, but I was afraid to cause any problem, put the name of a private company in title of the project

You can do as an example the ckb-next project, which only controls the corsair so why does not it load the name in the title? this led me to one such decision, but I may be wrong I needed to separate time to analyze

luizfschrickte, 2019-05-24:
You're right - they may have a reason!

OK, understood :)

Thank you!

Luiz Fernando Schrickte

rodgomesc, 2019-05-24:
anyway this would even help attract more developers and help users of laptops avell, I will talk to them in the chat and if not ouver problem I close this issue with the number of the authorization protocol and a print of the conversation with them

rodgomesc, 2019-05-24:
As discussed with an operator Ezequiel Tavares of the company avell of CNPJ: 06.787.479 / 0001-34, occurrence of protocol nº 201905241735744, the title of the project will become avell-unofficial-control-center, in order to reach a larger target audience, of Linux Users and  Developers, and thanks you @luizfschrickte for the suggestion

closed 2019-05-24 by rodgomesc

LeoniePhiline, 2019-06-08:
Since Avell is just of many resellers, I would propose another rename. ;)

Replace avell by tongfang-rgb? Or does anyone have better ideas?