rodgomesc/avell-unofficial-control-center #7: AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'

doulwyi, 2019-06-07:
aucc -c green -b 4
Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/aucc/main.py", line 86, in main
    control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
  File "/usr/local/lib/python2.7/dist-packages/aucc/main.py", line 37, in __init__
    super(ControlCenter, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python2.7/dist-packages/aucc/core/handler.py", line 50, in __init__
    super(DeviceHandler, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python2.7/dist-packages/aucc/core/handler.py", line 16, in __init__
    self._device = self._get_device(vendor_id, product_id)
  File "/usr/local/lib/python2.7/dist-packages/aucc/core/handler.py", line 27, in _get_device
    if device.is_kernel_driver_active(1):

hwinfo --short
cpu:                                                            
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3312 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2262 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2195 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 1943 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2888 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2841 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2743 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2813 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2780 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 1653 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2820 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3605 MHz
keyboard:
  /dev/input/event8    Maxxter Wireless Receiver
  /dev/input/event4    AT Translated Set 2 keyboard
mouse:
  /dev/input/mice      Maxxter Wireless Receiver
  /dev/input/mice      SynPS/2 Synaptics TouchPad
monitor:
                       N156HHE-GA1 LCD Monitor
                       AOC 2280W
graphics card:
                       nVidia GP107M [GeForce GTX 1050 Ti Mobile]
                       Intel UHD Graphics 630 (Mobile)
camera:
                       Apple iPhone5/5C/5S/6
sound:
                       Intel Cannon Lake PCH cAVS
storage:
                       Intel SATA controller
network:
  wlo1                 Intel Wireless-AC 9560 [Jefferson Peak]
  enp3s0f1             Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
network interface:
  enp3s0f1             Ethernet network interface
  docker0              Ethernet network interface
  enp0s20f0u1c4i2      Ethernet network interface
  wlo1                 Ethernet network interface
  lo                   Loopback network interface
disk:
  /dev/sda             WDC WDS250G2B0B-
partition:
  /dev/sda1            Partition
  /dev/sda2            Partition
usb controller:
                       Intel Cannon Lake PCH USB 3.1 xHCI Host Controller
bios:
                       BIOS
bridge:
                       Intel ISA bridge
                       Intel Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16)
                       Intel Cannon Lake PCH PCI Express Root Port #15
                       Intel 8th Gen Core Processor Host Bridge/DRAM Registers
                       Intel Cannon Lake PCH PCI Express Root Port #9
hub:
                       Linux Foundation 2.0 root hub
                       Linux Foundation 3.0 root hub
memory:
                       Main Memory
bluetooth:
                       Intel Bluetooth Device
unknown:
                       FPU
                       DMA controller
                       PIC
                       Keyboard controller
                       Realtek RTL8411B PCI Express Card Reader
                       Intel Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
                       Intel Cannon Lake PCH HECI Controller
                       Intel Cannon Lake PCH SPI Controller
                       Intel Cannon Lake PCH Thermal Controller
                       Intel Cannon Lake PCH Shared SRAM
                       Intel Cannon Lake PCH SMBus Controller
                       LighTuning EgisTec_ES603
                       Acer BisonCam, NB Pro
                       Apple iPhone5/5C/5S/6

rodgomesc, 2019-06-07:
what is the output for  `cat /sys/class/dmi/id/modalias `

doulwyi, 2019-06-07:
dmi:bvnAmericanMegatrendsInc.:bvr5.13:bd04/13/2018:svnNotebook:pnN8xEJEK:pvrNotApplicable:rvnNotebook:rnN8xEJEK:rvrNotApplicable:cvnNoEnclosure:ct10:cvrN/A:

rodgomesc, 2019-06-07:
> N8xEJEK is your clevo id

look 

https://bitbucket.org/tuxedocomputers/clevo-xsm-wmi/issues/53/support-for-clevo-n871ej

rodgomesc, 2019-06-07:
[image]

it seems that your laptop AVELL W1511 FOX, uses clevo products, thats use wmi (Windows Management Instrumentation Command-line), its could be hard to implement, so far I have no implementation plans, but as shown in the image above, [clevo-xsm-wmi](https://bitbucket.org/guimaluf/clevo-xsm-wmi/src/master/) can solve your problem.
first, you need to clone the branch that have your product support:

`git clone -b add_n8xejek_support https://bitbucket.org/guimaluf/clevo-xsm-wmi.git`

[here](https://bitbucket.org/guimaluf/clevo-xsm-wmi/src/master/) you can follow the installation instructions.
I will be closing this issue if there are any further questions I am available

closed 2019-06-07 by rodgomesc

invictus97k, 2021-01-07:
same issue
dmi:bvnAmericanMegatrendsInc.:bvrN.1.10PCS00:bd05/29/2020:br5.16:efr1.16:svnPCSpecialist:pnGK7NP5R:pvrStandard:rvnTongFang:rnGK7NP5R:rvrStandard:cvnStandard:ct10:cvrStandard:

raf-r, 2021-03-02:
I have the same issue.

dmi:bvnAmericanMegatrendsInc.:bvrN.1.15ESI01:bd08/10/2020:br5.16:efr1.22:svnGPUCompany:pnGWTN156-2BK:pvrStandard:rvnGPUCompany:rnGPUCompany:rvrStandard:cvnGPUCompany:ct10:cvrStandard: