rodgomesc/avell-unofficial-control-center #10: Advertise this wonderful project

LeoniePhiline, 2019-06-08:
These are follow-up ideas after reading #3.

1) The README should read all the different laptop models gaining such a huge improvement in linux support. This will help people find you via search engines.

- Tongfang GK5CN5Z / GK5CN6Z / GK5CQ7Z / GK5CP0Z (Barebone)
- Schenker XMG Neo 15 (German reseller)
- PCSpecialist Recoil II (UK reseller)
- Some OverPowered laptop model?
- Some Walmart laptop model.

2) Make the project known in this thread:
http://forum.notebookreview.com/threads/tongfang-gk5cn5z-gk5cn6z-gk5cq7z-gk5cp0z.815943/

3) Maybe get in contact with this dude who wrote an alternative control center for windows (and WIP for linux??) in C#:
http://forum.notebookreview.com/members/lkinder_bueno.704614/
Maybe there's some collaboration options?

4) Get in contact with XMG Schenker and let them add notes about linux support in their help sections and knowledge base:
- https://www.xmg.gg/
- https://bestware.com/en/

LeoniePhiline, 2019-06-08:
I sent an email to support@xmg.gg and created an account on http://forum.notebookreview.com/threads/tongfang-gk5cn5z-gk5cn6z-gk5cq7z-gk5cp0z.815943/ - but the account creation needs to be confirmed by an administrator first...

LeoniePhiline, 2019-06-09:
The message is out to http://forum.notebookreview.com/threads/tongfang-gk5cn5z-gk5cn6z-gk5cq7z-gk5cp0z.815943/page-430#post-10920858

It could also help people if we contact the other resellers (and Tongfang?!) to spread the word.

rodgomesc, 2019-06-09:
any progress?

astiak, 2019-08-06:
https://www.pcspecialist.co.uk/forums/threads/pcs-recoil-ii-iii-keyboard-backlight-control-in-linux.64211/

Posted a thread on the PCS forums.

rodgomesc, 2019-12-27:
@LeoniePhiline   they posted  [Here](http://forum.notebookreview.com/threads/tongfang-gk5cn5z-gk5cn6z-gk5cq7z-gk5cp0z.815943/)

search for @LeoniePhiline  in page content

phizeroth, 2020-06-21:
Confirmed this works on my Maingear Element, so it should work for all the new Intel white-label gaming laptops. They're pretty well-known so would be good to add to README:

- MAINGEAR Element
- XMG Fusion 15
- Eluktronics MAG-15