rodgomesc/avell-unofficial-control-center #8: `--style rainbow` switches the keyboard lights off

LeoniePhiline, 2019-06-08:
Hi,

first of all, thank you so much for developing this app!

I'm having an XMG Neo 15 = Tongfang GK5ZP0Z.
Almost all functions of the control center alpha work perfectly. 
However, the style `rainbow` is simply switching off the backlights. I tried passing a brightness parameter with it, but that did not help.

What can I try? Is there a chance the rainbow `light_style` should have a different array of values (`0x08, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00`) on my hardware? (All other stuff appears to be the same, as the CC works well - maybe there's different versions with slight differences? The XMG Neo 15 2019 version uses special "silent switches" - maybe the rgb controller got an update?)

Trying to debug with PyCharm, I get stuck at elevate(), where the connection to the debugger is dropped. How do you debug this? Are you using a local `sudo python` interpreter proxy script?
Sorry, I'm mostly doing ECMAScript and PHP - Python is something I'd like to get more acquainted with.

LeoniePhiline, 2019-06-08:
```
cat /sys/class/dmi/id/modalias                                                                                                                ✔  53% 🔋  18:14:40
dmi:bvnAmericanMegatrendsInc.:bvrN.1.51:bd05/07/2019:svnSchenkerTechnologiesGmbH:pnXMGNEO(M19,RTX2060):pvrStandard:rvnSchenkerTechnologiesGmbH:rnXMG_GK5CP0Z_GK7CP0S_M19:rvrStandard:cvnSchenkerTechnologiesGmbH:ct10:cvrStandard:
```

rodgomesc, 2019-06-08:
hmmm, looks like rainbow array of bytes is wrong

rodgomesc, 2019-06-08:
I'll fix it right now

> Hi,
>
> first of all, thank you so much for developing this app!
>
> I'm having an XMG Neo 15 = Tongfang GK5ZP0Z.
> Almost all functions of the control center alpha work perfectly.
> However, the style [...]

rodgomesc, 2019-06-08:
@LeoniePhiline i beleave it was fixed in https://github.com/rodgomesc/avell-unofficial-control-center/commit/146d7cab8fd5e74effb1c5b051581138ee4ff321, can you make new tests installing the project from this repo?

rodgomesc, 2019-06-08:
Looks like english is your Native language, can you help me checking If have Any errors in README? ir other place

LeoniePhiline, 2019-06-08:
@rodgomesc 

>  i beleave it was fixed in 146d7ca, can you make new tests installing the project from this repo?
I did a `git pull`, `python3 setup.py build`, `sudo python3 setup.py install` and tried `aucc -s rainbow` - and it worked perfectly.

Now it does show the rainbow - or at least what the manufacturer seems to see as a rainbow.
Do you have the same problem? The colors should be red, orange, yellow, green, teal, blue, purple - but they are in the wrong order: teal and blue are switched.
Is this a firmware-level issue, or can you fix that easily (before the single-key lighting is implemented)?

> Looks like english is your Native language, can you help me checking If have Any errors in README? ir other place
It's not my native language, but thank you for the compliment! :)
I will check and come back to you about that.

rodgomesc, 2019-06-08:
I did not understand very well what you meant by teal and blue are switched

LeoniePhiline, 2019-06-09:
> I did not understand very well what you meant by teal and blue are switched

Check out this image:
https://www.notebookcheck.net/fileadmin/Notebooks/Schenker/XMG_Neo_15_Turing/Schenker_XMG_Neo_15_Ansicht1.jpg

And this rainbow:
https://www.shutterstock.com/image-vector/color-rainbow-clouds-gradient-mesh-vector-597241106?src=H-yUYIIT5CmiZ88Fu1Z8Kg-1-1

On the laptop, the last four colors are:
green, **blue, teal (=turquoise=green-blue),** purple.

But blue and teal are in the wrong order.
The actual order of rainbow colors would be:

green, **teal (=turquoise=green-blue), then blue,** then purple.

So turqoise/teal is the color between green and blue, but the keyboard shows it between blue and purple - which is very annoying if you're someone who works with graphics. ;)

codelalala, 2019-10-23:
I want to report that with the most recent version, the rainbow style is not working. I tried all the styles, all others working correctly, but whenever I switched to rainbow the keyboard backlight turned completely off. Please let me know what kinds of logs might help.

Caffe1neAdd1ct, 2021-02-10:
Getting the same issue with a TongFang GK5CP6Z build, other styles work fine and static colours, rainbow turns off backlight completely.

Any testing or debugging I can try?

Caffe1neAdd1ct, 2021-03-11:
Installed with 1.0.3 via the manual method, the following styles error for me with:

```
Traceback (most recent call last):
  File "/usr/bin/aucc", line 33, in <module>
    sys.exit(load_entry_point('avell-unofficial-control-center==1.0.3', 'console_scripts', 'aucc')())
  File "/usr/lib/python3.9/site-packages/aucc/main.py", line 115, in main
    control.keyboard_style(parsed.style)
  File "/usr/lib/python3.9/site-packages/aucc/main.py", line 44, in keyboard_style
    self.ctrl_write(*light_style[style])
KeyError: 'wave'
```
```
sudo aucc -b 4 -s wave
sudo aucc -b 4 -s random
sudo aucc -b 4 -s breathing
sudo aucc -b 4 -s ripple
sudo aucc -b 4 -s reactiveripple
sudo aucc -b 4 -s reactiveaurora
sudo aucc -b 4 -s fireworks
```

Does not error but blank / no lights on keyboard:
sudo aucc -b 4 -s rainbow 

Working with correct colors / lights:
```
sudo aucc -b 4 -s marquee
sudo aucc -b 4 -s raindrop
sudo aucc -b 4 -s aurora
sudo aucc -b 4 -s reactive
```

Caffe1neAdd1ct, 2021-03-11:
Issue between seat and keyboard... pip had v1.0.1 installed, removed using pip uninstall and the manual installed 1.0.3 started being picked up and works fine :+1:  thanks for the help.