rodgomesc/avell-unofficial-control-center #13: Creating systemd integration to set desired color scheme at boot time

LeoniePhiline, 2019-06-09:
The default blue backlight colors are loaded on boot. **Any changes applied through the `aucc` command are not permanent.**

Since elevated privileges are necessary to communicate with the USB controller, placing an `aucc` script in your Desktop environment's start folder (e.g. `.config/autostart-scripts/`) is not comfortable, since you will be prompted for your `sudo` password each time your display manager starts up plasma or gnome (or whichever environment you prefer).

**Instead, the following should happen:**

- The GUI or command line tool writes the desired color scheme (or `aucc` parameters) into a system configuration file, like `/etc/aucc.conf`.
- A `systemd` service is created and enabled for auto-start during boot, which runs `aucc` with root privileges, applying the color scheme from `/etc/aucc.conf` to the USB keyboard controller.

rodgomesc, 2019-06-09:
I believe the most comfortable way to do this is using  crontab instead of systemd, as it would not be necessary to create a file either by simply typing` crontab -e` with elevated privileges, and adding:

`@reboot aucc -c color -b brightness`

this may be easier to understand for new users on linux

LeoniePhiline, 2019-06-09:
Do you know how `@reboot` handles suspend and hibernate?

Just now I tested hibernation (without crontab), and unfortunately the background lights do not 'survive' the hibernate-resume cycle. They are back to the default blue after resume.

rodgomesc, 2019-06-10:
hmmm, have you tried using systemD with timers? I'll do some tests

rodgomesc, 2019-06-10:
when suspending the leds return to normal for me by pressing any key, already while hibernating, hibernation does not work very well on my laptop with fedora 30 silverblue, I believe it is some issue

rodgomesc, 2019-06-10:
@LeoniePhiline 

**I do not know if you received my feedback on the donation, but I'll leave it anyway:**

[image]

PR #34 (2019-12-27): fix #13 #28 - now color changes are kept after reboot without additional configurations
fix #13 
fix #28 

now current color are kept after reboot by default,
commit 371587e6 (2019-12-27): Merge pull request #34 from rodgomesc/fix28

fix #13 #28 - now color changes are kept after reboot without additional configurations
closed 2019-12-27 by rodgomesc