rodgomesc/avell-unofficial-control-center #14: Add support for Avell C73 (Clevo PA70ES)

alcos0, 2019-06-20:
Hello.
I'd like to request support for the laptop Avell C73. What info I need to provide to make this possible?

`
cat /sys/class/dmi/id/modalias
dmi:bvnAmericanMegatrendsInc.:bvr1.07.08:bd08/23/2018:svnNotebook:pnC73:pvrNotApplicable:rvnNotebook:rnPA70ES:rvrNotApplicable:cvnNoEnclosure:ct10:cvrN/A:
`

And when I try to use aucc I get this error: ` AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active' `

Thanks.

rodgomesc, 2019-06-20:
Sorry, this driver only support laptops that use ITE Device(8291), If `sudo hwinfo --short` don't show something like;/:

```bash
keyboard:
                       Integrated Technology Express ITE Device(8291)
  /dev/input/event0    AT Translated Set 2 keyboard
```

Your Device is not conpatible, clevo is totally incompatible for now.

alcos0, 2019-06-21:
Hello. Thanks for replying.
`sudo hwinfo --short` actually shows this on the keyboard section:
```
keyboard:
  /dev/input/event9    Razer Orochi mouse
  /dev/input/event4    AT Translated Set 2 keyboard
```
Is it possible to add support for it?

rodgomesc, 2019-06-23:
Sorry but, Clevo devices and ITE devices are tottaly diferent, while ITE uses USB protocolo, Clevo use WMI, and i not have Clevo hardware to start a reverse Engineering and test to start a inicial support for It, 
so unless somebody donates some time in the future or trust to send me a device to start the tests, I can not do this implementation, maybe Avell will do it someday if you have plans to help with Linux compatibility

alcos0, 2019-06-24:
Alright, thanks!

closed 2019-06-24 by alcos0