rodgomesc/avell-unofficial-control-center #17: Keys turn off after setting color

vertexportus, 2019-07-03:
model: Avell G1550 FOX

when trying to set a set color, keys light up with selected color for about 1 second, and turn black;
Some styles do nothing (like rainbow), while others do work for a while, but afterwards turn off until a keypress on keyboard;

rodgomesc, 2019-07-03:
Thanks for your feedback, what is the Full output  of `sudo hwinfo --short`

rodgomesc, 2019-07-03:
How you installed this project,? Using pip or manually cloning this repo

vertexportus, 2019-07-03:
installed with pip directly from the repo

```
cpu:
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2592 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2000 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2315 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2400 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2383 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 1811 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2245 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2156 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2030 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2275 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2375 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2780 MHz
keyboard:
                       Integrated Technology Express ITE Device(8291)
  /dev/input/event9    Holtek USB Keyboard
  /dev/input/event4    AT Translated Set 2 keyboard
  /dev/input/event27   6C:5A:B5:48:1B:4D
mouse:
  /dev/input/mice      Razer Naga 2014
  /dev/input/event11   Holtek USB Keyboard
  /dev/input/mice      UNIW0001:00 093A:1336 Touchpad
monitor:
                       SAMSUNG S22E310
                       BOE HF LCD Monitor
graphics card:
                       nVidia GP106M [GeForce GTX 1060 Mobile]
                       Intel VGA compatible controller
sound:
                       Intel Audio device
storage:
                       Samsung Electronics Non-Volatile memory controller
                       Intel SATA controller
network:
  wlo1                 Intel WLAN controller
  enp4s0               Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
network interface:
  vethb81efcd          Ethernet network interface
  vethf5770f5          Ethernet network interface
  docker0              Ethernet network interface
  enp4s0               Ethernet network interface
  wlo1                 Ethernet network interface
  veth7cb9738          Ethernet network interface
  veth7efcfa2          Ethernet network interface
  veth5a0d1f4          Ethernet network interface
  lo                   Loopback network interface
  veth5dcffaa          Ethernet network interface
  vethd5cad41          Ethernet network interface
  br-db79a8d0553e      Ethernet network interface
  veth47c0922          Ethernet network interface
disk:
  /dev/nvme0n1         Samsung Electronics Disk
partition:
  /dev/nvme0n1p1       Partition
  /dev/nvme0n1p2       Partition
usb controller:
                       Intel USB Controller
bios:
                       BIOS
bridge:
                       Intel ISA bridge
                       Intel Skylake PCIe Controller (x16)
                       Intel PCI bridge
                       Intel Host bridge
                       Intel PCI bridge
                       Intel PCI bridge
hub:
                       Linux Foundation 2.0 root hub
                       Linux Foundation 3.0 root hub
memory:
                       Main Memory
bluetooth:
                       Intel Bluetooth Device
unknown:
                       FPU
                       DMA controller
                       PIC
                       Keyboard controller
                       PS/2 Controller
                       Intel Skylake Gaussian Mixture Model
                       Intel Communication controller
                       Intel Serial bus controller
                       Intel Communication controller
                       Intel Signal processing controller
                       Intel Serial bus controller
                       Intel Serial bus controller
                       Intel RAM memory
                       Intel SMBus
                       Realtek RTS5129 Card Reader Controller
                       Acer HD Webcam
  /dev/input/event17   Razer Naga 2014
                       Integrated Technology Express ITE Device(8291)

```

rodgomesc, 2019-07-03:
About rainbow style, is a but that i have fixed in 

https://github.com/rodgomesc/avell-unofficial-control-center/commit/146d7cab8fd5e74effb1c5b051581138ee4ff321

But not send to pypi yet ok

rodgomesc, 2019-07-03:
You have a mechanical keyboard? can send the command you are using , and not work?

vertexportus, 2019-07-03:
yes it's a mechanical keyboard, and I did install with pip3 but using the repo

`pip install git+https://github.com/rodgomesc/avell-unofficial-control-center.git`

it turns off after a few minutes, if I press a key it turns on again

LeoniePhiline, 2019-07-03:
The keyboard lighting turning off after a while of not typing is, AFAIK, a feature managed by the BIOS/EC, and out of control of `aucc`.

vertexportus, 2019-07-03:
weird though, because if I do not run `aucc` and leave the default lighting, it stays on forever

rodgomesc, 2019-07-04:
> weird though, because if I do not run `aucc` and leave the default lighting, it stays on forever

this is because when ITE 8192 don't recognize Any driver, by default BLUE color is keeps Active, is a hardware/firmware pattern

rodgomesc, 2019-07-09:
since have a few days without activity I will close the issue

closed 2019-07-09 by rodgomesc