rodgomesc/avell-unofficial-control-center #28: add aucc to run at startuup doesnt work

OMGdaDPS, 2019-11-05:
just as the title states, it doesnt seem to work for me. after i create the config file, and then go to chmod 77 the aucc.service, all i get is this error:
`dps@vulcan:~$ chmod 777 aucc.service
chmod: cannot access 'aucc.service': No such file or directory`

i installed this via the "easy" method, with pip3. i tried finding the actual package in my system (avell-unofficial-control-center), with pip3 show, and it returned a result that says that its in my /usr/local/lib/python3.6/dist-packages, but when i go to look in that directory, there isnt even a dist-packages directory lol.
  i dont know if that helps or not, 
 hope this reaches you well.

ricoaugo, 2019-11-05:
Try using CHMOD 755 and execute/create all the commands/files as root or sudo. In your printed error you are running as user "dps".

--
Ubuntu 18.04.03
Avell G1550 Fox 144hz
Core i7 9750h
64GB RAM
GTX 1060ti
2x Force MP300

OMGdaDPS, 2019-11-05:
ill try that as soon as i get home. thank you for responding, and thanks for making this great software. its so hard to even find support for some of these weirdly branded laptops, that almost all look the same. i cant even find a program on windows that would support the rgb keyboard.

ricoaugo, 2019-11-05:
@OMGdaDPS , I'm not the software developer. You need to thank @rodgomesc for it. I'm just an user from Brazil with the same issue in another branded laptop. Hope my experience can help you, that's the real purpose of open source software and collaboration.

OMGdaDPS, 2019-11-05:
OH! My apologies. At any rate, thanks for helping out!

rodgomesc, 2019-11-05:
it's worked ?

OMGdaDPS, 2019-11-07:
so i reinstalled the entire thing, using the pip3 install method, and running under sudo su (sudo), but i still get the error that i posted in the original post. i tried to use chmod 755 too, but it just seems like there is no aucc.service anywhere.

OMGdaDPS, 2019-11-07:
wait a second, i found it. i ran sudo systemctl status aucc.service, and it shows that its in /etc/systemd/system/aucc.service. but its disabled. should i move into the directory and run the chmod command?

ricoaugo, 2019-11-07:
I look again in your first post, it looks like you are running commands in the present folder not on /etc/systemd/system/aucc.service.
Try: chmod 755 /etc/systemd/system/aucc.service
And then
systemctl enable aucc.service

ricoaugo, 2019-11-07:
If it works, then finish with:
systemctl daemon-reload
systemctl start aucc.service
systemctl status aucc.service # to check if is running

OMGdaDPS, 2019-11-07:
yeah, i had no idea that i was supposed to be  in that directory, and THEN run the commands lol. i just got it to work! did a reboot and the config file loaded ! Nice! thank you guys!
Maybe we should ( i mean the dev of course) add a little more clarity, maybe just specifically state that you need to cd into that directory, and then run the chmod command. because people like me (relatively new to Linux) wont realize it.
  All in all, NICE!

OMGdaDPS, 2019-11-07:
i couldnt for the life of me find aucc.service lol. then i thought "hey, were using systemd to create a job or whatever, i bet it could tell me the status of it and thus show me where it is :) "

OMGdaDPS, 2019-11-07:
i know that it says to create the config file in the directory, but people may not get the hint about needing to actually BE in that directory to grant perms (like i didnt lol)

ricoaugo, 2019-11-07:
Nice. It's a good idea to clarify this on the steps @rodgomesc .
@OMGdaDPS just for the record and your future needs, you don't need actually be in the directory, you can also specify the entire path such as "chmod 755 /etc/systemd/system/aucc.service" This tricks works for any commands you need. Glad to help.

OMGdaDPS, 2019-11-07:
right on thank you!

Procyon-Lotor-Hub, 2019-11-15:
Hi Rodgomesc, great piece of software! i can confirm that it works on intel-tongfang QC7 (AKA xmg fusion 15 / eluktronics mag 15 / vapor pro 15). 

However i am having some rudimentary issues getting it to work with systemctl. 

Having used a text editor (kate) instead of VIM, as it drove me nuts, to generate the service file i run into the following issue.

> $ systemctl status aucc.service

 aucc.service - Avell Keyboard Configuration
   Loaded: loaded (/etc/systemd/system/aucc.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Fri 2019-11-15 19:40:13 CET; 35s ago
  Process: 10331 ExecStart=/usr/local/bin/aucc -c blue -b 1 (code=exited, status=203/EXEC)
 Main PID: 10331 (code=exited, status=203/EXEC)

 Nov 15 19:40:13 elliot systemd[1]: Starting Avell Keyboard Configuration...
 Nov 15 19:40:13 elliot systemd[10331]: aucc.service: Failed to execute command: No such file or directory
 Nov 15 19:40:13 elliot systemd[10331]: aucc.service: Failed at step EXEC spawning /usr/local/bin/aucc: No such file or directory
 Nov 15 19:40:13 elliot systemd[1]: aucc.service: Main process exited, code=exited, status=203/EXEC
 Nov 15 19:40:13 elliot systemd[1]: aucc.service: Failed with result 'exit-code'.
 Nov 15 19:40:13 elliot systemd[1]: Failed to start Avell Keyboard Configuration.

using sudo chmod 755 /etc/systemd/system/aucc.service works fine.

The kate generated .service file contains the following:
[Unit]
Description=Avell Keyboard Configuration
[Service]
Type=oneshot
ExecStart=/usr/local/bin/aucc -c blue -b 1
[Install]
WantedBy=multi-user.target

What do?

PR #34 (2019-12-27): fix #13 #28 - now color changes are kept after reboot without additional configurations
fix #13 
fix #28 

now current color are kept after reboot by default,
closed 2019-12-27 by rodgomesc