rodgomesc/avell-unofficial-control-center #29: setting certain "styles", ends in no lights

OMGdaDPS, 2019-11-05:
when i try to set the "rainbow" or "reactive" preset styles, it doesn't do anything, and then when i try to set it to something regular like a solid color it doesn't do anything. I end up having to reboot my system to get it to work again.

rodgomesc, 2019-11-05:
what is the complete output for `sudo lsusb -d 048d:ce00 -vvv`

OMGdaDPS, 2019-11-07:
`dps@vulcan:~$ sudo lsusb -d 048d:ce00 -vvv
[sudo] password for dps: 

Bus 001 Device 004: ID 048d:ce00 Integrated Technology Express, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x048d Integrated Technology Express, Inc.
  idProduct          0xce00 
  bcdDevice            0.01
  iManufacturer           1 ITE Tech. Inc.
  iProduct                2 ITE Device(8291)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      23
          Report Descriptor: (length is 23)
            Item(Global): Usage Page, data= [ 0x89 0xff ] 65417
                            (null)
            Item(Local ): Usage, data= [ 0x10 ] 16
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Report ID, data= [ 0x5a ] 90
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x10 ] 16
            Item(Main  ): Feature, data= [ 0x00 ] 0
                            Data Array Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      31
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered
`

OMGdaDPS, 2019-11-07:
i tried to set that in code format, but it keeps doing that for some reason

rodgomesc, 2019-11-07:
the poblem is... your device I.T.E 8291, are in revision, `0.01` and this project is only compatible with revision `0.03`, check [README](https://github.com/rodgomesc/avell-unofficial-control-center/blob/master/README.md)

OMGdaDPS, 2019-11-08:
hmmm, i see. you know, im able to control it in the bios. there are a few settings or "styles" that i can set it to, but they dont stay after i boot into something. theyre pretty good, and im wondering if there is an "upstream" utility, or some kind of sdk that goes along with this i.t.e 8291. do you know of anything like that? maybe i could start working on support for my version..

OMGdaDPS, 2019-11-08:
at least everything else seems to work. like setting it to the h and v flags, and the different colors. and a couple of the styes do work too.

OMGdaDPS, 2019-11-08:
maybe, i was thinking... could it be that mine has a smaller eFlash than the newer revisions? i cant seem to find this specific model (8291) on their products page, but i did find the newer IT8296e version. their are two revisions for this one as well, one with 56k of flash and the other has 120k. hmm, idk.
  This is actually why i came to linux: hardware. i love hardware.im studying microcontrollers and electronics right now, but im not that well versed so far. i can upload a blink sketch, and rip firmware from spi flash chips lmao.
  if i can be of any help, please let me know

wifiuk, 2019-11-13:
yeah i think i have 0.01 too

rodgomesc, 2019-11-18:
Sorry but, since i dont have ite8291 revision 0.1 in my hands i can't support It, there is no way to analyse and start a reverse enginer

closed 2019-11-18 by rodgomesc