rodgomesc/avell-unofficial-control-center #31: Cant run color command

Se7Touch, 2019-11-09:
My problem in attachment
[image]

rodgomesc, 2019-11-10:
sound's like you not have libusb in your system, what is your SO?

rodgomesc, 2019-11-10:
Try run `brew install libusb`

rodgomesc, 2019-11-10:
Check [this](https://github.com/rodgomesc/avell-unofficial-control-center/blob/master/aucc/tests/main.py) code, for a while only 2 colors Will work on macOS

Se7Touch, 2019-11-10:
My laptop model is your laptop model turkish mirror :)

TR Seller Monster

Monster Abra A7 v10.1

17.3 inch
i7 8750h
1050
8gb/256gb
Hm370 chipset

Se7Touch, 2019-11-10:
i will try your method after one hour ago ty. İm back you...

rodgomesc, 2019-11-11:
> i will try your method after one hour ago ty. İm back you...

did work ?

Se7Touch, 2019-11-11:
unfortunately it did not work. I'm using kirainmoe's star beat software. you just get used to the rainbow mode.

rodgomesc, 2019-11-11:
Do you install hidapi?

Se7Touch, 2019-11-11:
But im a simple pc user, I don't understand much. :)

rodgomesc, 2019-11-12:
did you got hdmi working on you hackintosh?

Se7Touch, 2019-11-12:
No, because the HDMI port comes out of the external video card. work on you hackintosh ? In the meantime, our devices are exactly the same. Our only difference is the brand. brand monster in my country. yours avell

rodgomesc, 2019-11-12:
Do you have Any issues with hackintosh?, HDMI dont work for me too

Se7Touch, 2019-11-12:
Hmm.

Working:
1. Keyboard Light (Only Rainbow)
2. Sleep Mode Perfectly with darkwake=8
3. Sound AppleALC.kext inject 88 layout id
4. Power Management Perfectly with HWPenabler.kext and CPUFriend.kext (VirtualSCM)
5. 2GB Intel UHD 630
6. Touchpad with multitouch gestures
7. FN Keys (Radio on/off F4 - Adjust keyboard light F6, F7 - Sound F8, F9, F10 - Adjust brightness F11, F12)
8. Battery Percent
9. Night Shift
10. Usb 3/2 ports

Not Working:
1. Bluetooth/Wifi (I bought Dell Wifi Card its coming from China)
2. Some keyboard light (color/mode)
3. My CPU heating always minimum 38 - 45 celcius that normal ? i don't know it...

rodgomesc, 2019-11-12:
Can you send me your efi folder?, I can help you change keyboard lights to
Green or Red,

Your cpu turbo boost are working?
And i can help fix cpu turbo boost If is not working

rodgomesc, 2019-11-12:
So let's make keyboard lights working, First download [this script](https://raw.githubusercontent.com/rodgomesc/avell-unofficial-control-center/master/aucc/tests/main.py) and run the commands of first two lines, after try `sudo python main.py`

Se7Touch, 2019-11-12:
[Se7_EFI.zip](https://github.com/rodgomesc/avell-unofficial-control-center/files/3837752/Se7_EFI.zip)

Your cpu turbo boost are working? > probably
And i can help fix cpu turbo boost If is not working > How can i test ?

rodgomesc, 2019-11-13:
Download Intel Power gadget, try stress your cpu and look If cpu clock comes to 4.1Ghz

Se7Touch, 2019-11-13:
Yep its working...

I found a problem in my test yesterday. Short sleep mode works perfectly. However, when I tried the long sleep mode, after a while I woke up the black screen. Then I put the device to sleep before going to bed at night. And when I woke up at night I realized that the device had turned itself on. And it was working with a black screen. do you have any idea what caused this problem?

rodgomesc, 2019-11-18:
> So let's make keyboard lights working, First download [this script](https://raw.githubusercontent.com/rodgomesc/avell-unofficial-control-center/master/aucc/tests/main.py) and run the commands of first two lines, after try `sudo python main.py`

This worked for you?

Se7Touch, 2019-11-20:
Not work...

I update my EFI now fully working with sleep mode.

https://yadi.sk/d/6ODuNLoC7wI-Sg

Se7Touch, 2019-11-25:
> So let's make keyboard lights working, First download [this script](https://raw.githubusercontent.com/rodgomesc/avell-unofficial-control-center/master/aucc/tests/main.py) and run the commands of first two lines, after try `sudo python main.py`

Its work. Ty.

rodgomesc, 2019-12-02:
> So let's make keyboard lights working, First download [this script](https://raw.githubusercontent.com/rodgomesc/avell-unofficial-control-center/master/aucc/tests/main.py) and run the commands of first two lines, after try `sudo python main.py`

cool, i will close this, let me now if you have any issues, thanks for efi, it works like a charm

closed 2019-12-02 by rodgomesc

Se7Touch, 2020-05-30:
> So let's make keyboard lights working, First download [this script](https://raw.githubusercontent.com/rodgomesc/avell-unofficial-control-center/master/aucc/tests/main.py) and run the commands of first two lines, after try `sudo python main.py`

Hello,

I need this script again bro, lost my files :/ And I have too problem running this tool. :D