rodgomesc/avell-unofficial-control-center #35: Changes are not saving

astiak, 2020-01-11:
Hi,

I haven't had time to fully investigate this yet, but from the little I have done it appears as though the option to save changes is not functioning on my laptop.

Machine:
```
PC Specialist product: Recoil III
TongFang model: GK7CP7S
UEFI BIOS: American Megatrends v: N.1.03
```
Keyboard model:
```
Keyboard:
  /dev/input/event21   MosArt 2.4G Keyboard Mouse
                       Integrated Technology Express ITE Device(8291)
  /dev/input/event5    AT Translated Set 2 keyboard
```

My BIOS settings have options for other styles to start in, but I've tried to set this to "disabled", and it defaults back to the wave style, which is what ends up sticking after a reboot. I've tried setting -V, -H and various named styles, none of which persist after a reboot. It is entirely possible that my BIOS is the issue here.

I don't know if anyone else is experiencing a similar issue. When I do get time I'll attempt to do some more digging.

rodgomesc, 2020-01-11:
For a while this only Works with colors defined using -c,

Example:

aucc -c red -b 4

rodgomesc, 2020-01-11:
What is your ITE revision?

astiak, 2020-01-11:
```
  Device: usb 0xce00 "ITE Device(8291)"
  Revision: "0.03"
```

astiak, 2020-01-13:
> For a while this only Works with colors defined using -c,
> 
> Example:
> 
> aucc -c red -b 4

I've also tried this, but it automatically resets to wave after reboot.

michael-schaefer-eu, 2020-02-07:
Same problem on ubuntu 19.10

I just installed rc.local script to execute on startup:

```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

aucc -c teal

exit 0
```

badmannen, 2020-02-09:
It's pretty useful to put this script also to execute before and after sleep

astiak, 2020-05-09:
@michael-schaefer-eu 

I've used the same approach, just odd that it saves for some and not others. Still investigating why this is the case.

michael-schaefer-eu, 2020-05-15:
@astiak  ¯\_(ツ)_/¯