rodgomesc/avell-unofficial-control-center #39: No module named USB, No module name elevate

OMGdaDPS, 2020-02-02:
After installing, this is the output:

`dps@vulcan:~$ sudo pip3 install avell-unofficial-control-center
Collecting avell-unofficial-control-center
  Using cached https://files.pythonhosted.org/packages/e1/e6/99ec1af7a362344849cea9cda71c115b54f2d963dce42ff66ecf3700e5e5/avell_unofficial_control_center-1.0.1-py3-none-any.whl
Requirement already satisfied: elevate in /root/.local/lib/python3.7/site-packages (from avell-unofficial-control-center) (0.1.3)
Requirement already satisfied: pyusb in /root/.local/lib/python3.7/site-packages (from avell-unofficial-control-center) (1.0.2)
Installing collected packages: avell-unofficial-control-center
Successfully installed avell-unofficial-control-center-1.0.1
dps@vulcan:~$ 
`

It tells me that the requirements are satisfied, but when I go to run `aucc`, I am presented with this error:

`Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 6, in <module>
    from aucc.main import main
  File "/usr/local/lib/python3.7/dist-packages/aucc/main.py", line 10, in <module>
    from aucc.core.handler import DeviceHandler
  File "/usr/local/lib/python3.7/dist-packages/aucc/core/handler.py", line 8, in <module>
    import usb.core
ModuleNotFoundError: No module named 'usb'`

This is after running a normal command, like `aucc -H blue white -b 4`, then it asks me for my `sudo` password, and then am hit with this error.

I would like to point out that your program has always worked, until I just did a fresh install of Ubuntu 19.10 (which is the same release it was running before) and had to reinstall everything.

Any idea of what it could be?

OMGdaDPS, 2020-02-02:
Alright, so I figured something out. I am still not sure WHY it's like this, this time, but I got it to work just by using `sudo` before the command. 

Before though, I could just do `aucc -H white blue -b 4` or whatever command, without `sudo` and the desktop would show a prompt, prompting me for my `sudo` password. Any ideas why?

joapuiib, 2020-02-03:
I think the elevate module on python does that. (https://pypi.org/project/elevate/).
It detects if the program is executed as root and, if not, asks for root password and relaunches it.

```python
def main():
    from elevate import elevate

    if not os.geteuid() == 0:
        elevate()
```

OMGdaDPS, 2020-02-05:
Just did a fresh install of Ubuntu 19.10, and installed it the exact same way as I always do, and it works how I am expecting it to. I wonder what could have happened the last time? 
  I mean, I kept uninstalling it and then reinstalling it, and it would produce the same exact result (having to type `sudo` before `aucc` to get it to work). 
  Yeah, a bit confusing to me. I wish that I knew more about software development so I could try and debug what happened.