rodgomesc/avell-unofficial-control-center #40: [Question] mapping the Fn key

rkitover, 2020-02-10:
Hi, thank you for this great project.

I had a question about these keyboards, and since you are more intimately familiar with them perhaps you could steer me in the right direction.

On Windows the Fn key maps to a key code that autohotkey can make mappings with.

For example, I have this to make Fn + Down be my media play/pause key:

https://gist.github.com/rkitover/0d9cd84b93f873c13034196c132d7d4f

when I use `xev` in gnome to see if there is any reaction to the Fn key on linux, there is none.

I wonder where I start in the layers of the linux desktop to generate a key code for this key on this hardware.

Also just tried something else, I did a `cat /dev/input/event0` which is the device node for the keyboard on my laptop, and pressing the Fn key **DID** generate activity from the device node.

Any suggestions greatly appreciated!

pobrn, 2020-07-14:
*I hope it is still relevant:*

Try using `evtest` to see if a scan code is recognised when you press `Fn`, then following the guide [here](https://wiki.archlinux.org/index.php/Map_scancodes_to_keycodes).

rkitover, 2020-07-14:
Thank you, I will try that!

rkitover, 2021-06-08:
By the way, I packaged this software for nix (and nixos), under the package name avell-unofficial-control-center.

rkitover, 2021-06-22:
So I finally got something working, this is the crappy initial version:

https://github.com/rkitover/fnkeyd