rodgomesc/avell-unofficial-control-center #43: Does not appear the: ITE device (8291) on keyboard:...

ricardoifc, 2020-04-28:
I uninstalled Hackintosh on my laptop and after that 
the ITE Device (8291) disappeared in hwinfo
I installed aucc and it doesn't work, In windows it doesn't work either :(
**can you help me enable my rgb keyboard? please**

### **hwinfo**
ricardo@ricardoifc:~$ hwinfo --short
cpu:                                                            
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3800 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3813 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3318 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3812 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3560 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3893 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3916 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3770 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3938 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3855 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3933 MHz
                       Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 3666 MHz
keyboard:
  /dev/input/event4    AT Translated Set 2 keyboard
mouse:
  /dev/input/mice      SiGma Micro Usb Mouse
  /dev/input/mice      UNIW0001:00 093A:0003 Touchpad
monitor:
                       LCD Monitor
graphics card:
                       nVidia GP107M [GeForce GTX 1050 Ti Mobile]
                       Intel UHD Graphics 630 (Mobile)
sound:
                       Intel Cannon Lake PCH cAVS
                       nVidia GP107GL High Definition Audio Controller
storage:
                       Intel Cannon Lake Mobile PCH SATA AHCI Controller
network:
  wlo1                 Intel Wireless-AC 9560 [Jefferson Peak]
  enp4s0f1             Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
network interface:
  lo                   Loopback network interface
  enp4s0f1             Ethernet network interface
  wlo1                 Ethernet network interface
disk:
  /dev/sda             WDC WDS240G2G0B-
partition:
  /dev/sda1            Partition
  /dev/sda2            Partition
  /dev/sda3            Partition
  /dev/sda4            Partition
  /dev/sda5            Partition
usb controller:
                       Intel Cannon Lake PCH USB 3.1 xHCI Host Controller
bios:
                       BIOS
bridge:
                       Intel ISA bridge
                       Intel Cannon Lake PCH PCI Express Root Port #21
                       Intel Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16)
                       Intel Cannon Lake PCH PCI Express Root Port #17
                       Intel 8th Gen Core Processor Host Bridge/DRAM Registers
                       Intel Cannon Lake PCH PCI Express Root Port #15
hub:
                       Linux Foundation 2.0 root hub
                       Linux Foundation 3.0 root hub
memory:
                       Main Memory
unknown:
                       FPU
                       DMA controller
                       PIC
                       Keyboard controller
                       Intel Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
                       Intel Cannon Lake PCH HECI Controller
                       Intel Cannon Lake PCH SPI Controller
                       Intel Communication controller
                       Intel Cannon Lake PCH Thermal Controller
                       Intel Cannon Lake PCH Serial IO I2C Controller #0
                       Intel Cannon Lake PCH Shared SRAM
                       Realtek RTL8411B PCI Express Card Reader
                       Intel Cannon Lake PCH SMBus Controller
                       Chicony Electronics HD Webcam
                       Integrated Technology Express ITE Upgrade Mode

AdsonCicilioti, 2020-09-25:
Here also not appear.

AdsonCicilioti, 2020-09-25:
My Avell is an LIV52 2020 model