rodgomesc/avell-unofficial-control-center #44: AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'

caioaamaral, 2020-05-07:
Tried at avell G1544 FOX, any `aucc` command gives me the following error (also it keeps asking for sudo)
```console
aucc -c green -b 4
Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/aucc/main.py", line 86, in main
    control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
  File "/usr/local/lib/python3.5/dist-packages/aucc/main.py", line 37, in __init__
    super(ControlCenter, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python3.5/dist-packages/aucc/core/handler.py", line 50, in __init__
    super(DeviceHandler, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python3.5/dist-packages/aucc/core/handler.py", line 16, in __init__
    self._device = self._get_device(vendor_id, product_id)
  File "/usr/local/lib/python3.5/dist-packages/aucc/core/handler.py", line 27, in _get_device
    if device.is_kernel_driver_active(1):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
```

caioaamaral, 2020-05-08:
My bad, seems like hwinfo does not show `Integrated Technology Express ITE Device(8291)`  in `keyboard` section. So G1544 FOX devices are not compatible, you may close this issue tks

thotty, 2021-02-07:
see #48 :)