rodgomesc/avell-unofficial-control-center #45: Using udev rather than elevate

astiak, 2020-05-09:
This is more a discussion than anything else.

Rather than typing sudo for every change, I was wondering if a udev rule would be more appropriate.

`sudo vim /etc/udev/rules.d/99-aucc.rules`
Add the following:
`SUBSYSTEM=="usb", ATTR{idVendor}=="048d", ATTR{idProduct}=="ce00", MODE="666"`
Save, exit, reboot.

I've commented out the elevate section of main.py and haven't seen any issues thus far. I'm able to choose styles and colours without having to input a password every time.

The only problem of course is getting people to create the rule when installing aucc. I'm not well versed enough in python to know if this can be automated. Or perhaps a file can be included in the source with instructions added to the readme to `sudo cp` it to the correct location.

joapuiib, 2021-03-10:
I totally agree with this. Typing sudo every time is a bit annoying.

I'm not familiar with `udev`, so I can't give an opinion on that.

I thought about setting the SETUID permission to the program, but it seems that it's not allowed on interpreted scripts (See: [Setuid bit on python script](https://stackoverflow.com/questions/8314012/setuid-bit-on-python-script-linux-vs-solaris).
A possible solution would be make a c++ wrapper, just as suggested on the previous StackOverflow post.

joapuiib, 2021-03-15:
I managed to do it making a binary version of the python script with [PyInstaller](https://pypi.org/project/pyinstaller/) and using the SETUID permission.

I managed to do it with the following lines:

```bash
#!/bin/bash

pyinstaller -F -n aucc aucc/main.py

sudo chown root:root dist/aucc
sudo chmod u+s dist/aucc

AUCC_PATH=`which aucc`
sudo cp dist/aucc $AUCC_PATH
```