rodgomesc/avell-unofficial-control-center #48: AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'

TreeFrogApps, 2020-07-14:
Hi 

I have a TongFang GM5MPHW. After installing your control center I was presented with this error :

`Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/aucc/main.py", line 86, in main
    control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
  File "/usr/local/lib/python3.8/dist-packages/aucc/main.py", line 37, in __init__
    super(ControlCenter, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python3.8/dist-packages/aucc/core/handler.py", line 50, in __init__
    super(DeviceHandler, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python3.8/dist-packages/aucc/core/handler.py", line 16, in __init__
    self._device = self._get_device(vendor_id, product_id)
  File "/usr/local/lib/python3.8/dist-packages/aucc/core/handler.py", line 27, in _get_device
    if device.is_kernel_driver_active(1):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'`

Having checked `hwinfo --short` I get : 

    `keyboard:  Integrated Technology Express ITE Device(8291)`

Also checking `lshw` I get :

`*-usb:1 UNCLAIMED
                   description: Keyboard
                   product: ITE Device(8291)
                   vendor: ITE Tech. Inc.
                   physical id: 6
                   bus info: usb@1:6
                   version: 0.03
                   capabilities: usb-2.00
                   configuration: maxpower=100mA speed=12Mbit/s`

So it would appear that my keyboard is indeed compatible.

The answer lies in this error line :

`control = ControlCenter(vendor_id=0x048d, product_id=0xce00)`

Running `lsusb` shows :

`Bus 001 Device 003: ID 048d:6004 Integrated Technology Express, Inc. ITE Device(8291)`

Which as you can see has a different product id.  Once I knew this I looked into the `main.py` file and found that the product id - `0xce00` is hardcode, as soon as I changed this to `6004` then ran `aucc -c green` everything worked.

It would seem that there maybe a product id change for the manufacturer under the same product revision - which seems weird, but may account why other users might seem to have the correct hardware, but are getting this error.

robintalma, 2020-08-04:
hi, same laptop here, and same error. Please help :)

TreeFrogApps, 2020-08-04:
First clone the repo, then try :

`lsusb | grep "Integrated Technology Express"`

If anything like this appears like :

`Bus 001 Device 003: ID xxxx:xxxx Integrated Technology Express, Inc. ITE Device(8291)`

Note down the ID number, now look inside `main.py` specifically the main function. If your id you jotted down was anything other than : `048d:ce00` then change this line :

`control = ControlCenter(vendor_id=0x048d, product_id=0xce00)`

to :

`control = ControlCenter(vendor_id=0x[id values left of colon], product_id=0x[id values right of the colon])`

Then just follow the instructions on the main readme for building and installing from source :

Build an installable package: python3 setup.py build
Install the package: sudo python3 setup.py install

Now try running `sudo aucc -b 2 -c green` to test if it works.

robintalma, 2020-08-18:
`lsusb | grep "Integrated Technology Express"`

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1ea7:0064 SHARKOON Technologies GmbH 2.4G Mouse
Bus 001 Device 002: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 004: ID 8087:0026 Intel Corp. 
Bus 001 Device 003: ID 04f2:b641 Chicony Electronics Co., Ltd HD Webcam
Bus 001 Device 007: ID 0416:5020 Winbond Electronics Corp. HID Transfer
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```  
It looks like I don't have "Integrated Technology Express" device. 

My vendor (PCspecialist) tells me that my computer is a "TongFang GK7MP5R". On README.md some TongFang are supported, maybe mine is not ?

rodgomesc, 2020-08-18:
sound's like your computer uses another chip to control rgb keyboard

rodgomesc, 2020-08-18:
> Then just follow the instructions on the main readme for building and installing from source :

changing vendor id and product id is not solve anything, this code only works with ite.8291 rev03, each revision / model has a different set of instructions

douglasrizzo, 2020-10-04:
Hi everyone. I tried using the project on my newly bought Avell C62 LIV and also got this error. In my case, I did not get the line:

```
Integrated Technology Express ITE Device(8291)
```

I just got:

```
AT Translated Set 2 keyboard
```

I assume my laptop is not compatible at all, so I'm giving you guys a heads-up that not all Avell laptops may be compatible with this driver.

space-hunter, 2020-12-26:
got the same error 'AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'' 
after changing the product_id and reinstall it works, thanks

lsusb | grep "Integrated Technology Express"
Bus 001 Device 002: ID 048d:6004 Integrated Technology Express, Inc. ITE Device(8291)

old:	    control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
new	    control = ControlCenter(vendor_id=0x048d, product_id=0x6004)

Luux, 2021-03-01:
> Device: Schenker NEO 15 Build: 2020/12
> got the same error 'AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active''
> after changing the product_id and reinstall it works, thanks
> 
> lsusb | grep "Integrated Technology Express"
> Bus 001 Device 002: ID 048d:6004 Integrated Technology Express, Inc. ITE Device(8291)
> 
> old: control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
> new control = ControlCenter(vendor_id=0x048d, product_id=0x6004)

I tried it with the same type of notebook (and the same id as well), but the problem still persists.

SimoFiuz, 2021-04-29:
Same issue here on an XMG CORE 14. My keyboard is just backlit with no RGB.

`keyboard:
  /dev/input/event3    AT Translated Set 2 keyboard
`

lucas-mior, 2023-02-24:
@douglasrizzo have you found a solution? I'm having the same issue with avell.

douglasrizzo, 2023-02-24:
> @douglasrizzo have you found a solution? I'm having the same issue with avell.

Nope, gave up a long time ago!