rodgomesc/avell-unofficial-control-center #56: Recoil III (GK5CP6Z) LED Bar control

Caffe1neAdd1ct, 2021-02-10:
The PC Specialist recoil III has an LED bar across the front, would it be possible to add controls for this as well?

siphomateke, 2021-05-08:
@Caffe1neAdd1ct You may be interested in the application I developed for controlling the LED bar on the GK5CP0Z: https://gitlab.com/siphomateke/tongfang-control. It's still very much WIP, but I currently use it to control the LED bar on my personal laptop. avell-unofficial-control-center controls the keyboard's RGB using a USB driver whereas controlling the LED bar requires lower-level ACPI calls, so it's unlikely this repository will be gaining support for the LED bar any time soon.

Caffe1neAdd1ct, 2021-05-08:
Thanks @siphomateke I'll take a look that the other repo later, it looks great!

closed 2021-05-08 by Caffe1neAdd1ct