rodgomesc/avell-unofficial-control-center #60: is_kernel_driver_active

53jk1, 2021-07-27:
Hi i tried to install this on my PC, I have built-in keyboard in laptop, but i'm getting traceback like this.

Traceback
```
  File "/usr/local/bin/aucc", line 11, in <module>
    load_entry_point('avell-unofficial-control-center==1.0.4', 'console_scripts', 'aucc')()
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/main.py", line 165, in main
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/main.py", line 104, in __init__
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/core/handler.py", line 50, in __init__
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/core/handler.py", line 16, in __init__
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/core/handler.py", line 27, in _get_device
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'  File "/usr/local/bin/aucc", line 11, in <module>
    load_entry_point('avell-unofficial-control-center==1.0.4', 'console_scripts', 'aucc')()
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/main.py", line 165, in main
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/main.py", line 104, in __init__
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/core/handler.py", line 50, in __init__
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/core/handler.py", line 16, in __init__
  File "/usr/local/lib/python3.6/dist-packages/avell_unofficial_control_center-1.0.4-py3.6.egg/aucc/core/handler.py", line 27, in _get_device
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
```

busybeaver42, 2021-09-27:
I have the same issue:
Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 11, in <module>
    load_entry_point('avell-unofficial-control-center==1.0.4', 'console_scripts', 'aucc')()
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/main.py", line 165, in main
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/main.py", line 104, in __init__
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/core/handler.py", line 50, in __init__
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/core/handler.py", line 16, in __init__
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/core/handler.py", line 27, in _get_device
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
root@xpr15m19:/home/joerg/keyboard/avell-unofficial-control-center# aucc -s rainbow
Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 11, in <module>
    load_entry_point('avell-unofficial-control-center==1.0.4', 'console_scripts', 'aucc')()
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/main.py", line 165, in main
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/main.py", line 104, in __init__
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/core/handler.py", line 50, in __init__
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/core/handler.py", line 16, in __init__
  File "/usr/local/lib/python3.8/dist-packages/avell_unofficial_control_center-1.0.4-py3.8.egg/aucc/core/handler.py", line 27, in _get_device
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'

XMGpro15 [XPR15M19]

aadm, 2021-10-02:
Same here, laptop is a PC Specialist Recoil IV i7-10875H / Geforce RTX2070.

Used to work on Ubuntu 20.04 and 20.10, after upgrading to 21.04 `aucc` does not work anymore and gives me this error msg (same as user above):

```
(base) 19:56:47 aadm@psion:~$ sudo aucc -H pink teal -b 4
Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/aucc/main.py", line 86, in main
    control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
  File "/usr/local/lib/python3.9/dist-packages/aucc/main.py", line 37, in __init__
    super(ControlCenter, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python3.9/dist-packages/aucc/core/handler.py", line 50, in __init__
    super(DeviceHandler, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python3.9/dist-packages/aucc/core/handler.py", line 16, in __init__
    self._device = self._get_device(vendor_id, product_id)
  File "/usr/local/lib/python3.9/dist-packages/aucc/core/handler.py", line 27, in _get_device
    if device.is_kernel_driver_active(1):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
```

53jk1, 2021-11-07:
Thanks, I'll let my friend who I gave this laptop to take a look at it.