rodgomesc/avell-unofficial-control-center #63: AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'

carreirorco, 2021-11-08:
```
rafael@a52mob:~$ aucc 
Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 33, in <module>
    sys.exit(load_entry_point('avell-unofficial-control-center==1.0.4', 'console_scripts', 'aucc')())
  File "/usr/local/lib/python3.9/dist-packages/avell_unofficial_control_center-1.0.4-py3.9.egg/aucc/main.py", line 166, in main
  File "/usr/local/lib/python3.9/dist-packages/avell_unofficial_control_center-1.0.4-py3.9.egg/aucc/main.py", line 104, in __init__
  File "/usr/local/lib/python3.9/dist-packages/avell_unofficial_control_center-1.0.4-py3.9.egg/aucc/core/handler.py", line 50, in __init__
  File "/usr/local/lib/python3.9/dist-packages/avell_unofficial_control_center-1.0.4-py3.9.egg/aucc/core/handler.py", line 16, in __init__
  File "/usr/local/lib/python3.9/dist-packages/avell_unofficial_control_center-1.0.4-py3.9.egg/aucc/core/handler.py", line 27, in _get_device
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
```

```
rafael@a52mob:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 046d:c092 Logitech, Inc. G203 LIGHTSYNC Gaming Mouse
Bus 001 Device 003: ID 8087:0026 Intel Corp. 
Bus 001 Device 002: ID 04f2:b71a Chicony Electronics Co., Ltd HD Webcam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
rafael@a52mob:~$ sudo hwinfo --keyboard
29: PS/2 00.0: 10800 Keyboard                                   
  [Created at input.226]
  Unique ID: 9ui9.+49ps10DtUF
  Hardware Class: keyboard
  Model: "AT Translated Set 2 keyboard"
  Vendor: 0x0001 
  Device: 0x0001 "AT Translated Set 2 keyboard"
  Compatible to: int 0x0211 0x0001
  Device File: /dev/input/event3
  Device Files: /dev/input/event3, /dev/input/by-path/platform-i8042-serio-0-event-kbd
  Device Number: char 13:67
  Driver Info #0:
    XkbRules: xfree86
    XkbModel: pc104
  Config Status: cfg=new, avail=yes, need=no, active=unknown
rafael@a52mob:~$ 

```

aidanmcdonagh22, 2021-12-01:
Same error for me as well, my keyboard output:

`31: USB 00.0: 10800 Keyboard                                    
  [Created at usb.122]
  Unique ID: JAsG.is7I5T49lQ1
  Parent ID: uIhY.2DFUsyrieMD
  SysFS ID: /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0
  SysFS BusID: 3-11:1.0
  Hardware Class: keyboard
  Model: "Integrated Technology Express ITE Device(8291)"
  Hotplug: USB
  Vendor: usb 0x048d "Integrated Technology Express, Inc."
  Device: usb 0x6004 "ITE Device(8291)"
  Revision: "0.03"
  Speed: 12 Mbps
  Module Alias: "usb:v048Dp6004d0003dc00dsc00dp00ic03isc01ip01in00"
  Driver Info #0:
    XkbRules: xfree86
    XkbModel: pc104
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #25 (Hub)

36: PS/2 00.0: 10800 Keyboard
  [Created at input.226]
  Unique ID: 9ui9.+49ps10DtUF
  Hardware Class: keyboard
  Model: "AT Translated Set 2 keyboard"
  Vendor: 0x0001 
  Device: 0x0001 "AT Translated Set 2 keyboard"
  Compatible to: int 0x0211 0x0001
  Device File: /dev/input/event3
  Device Files: /dev/input/event3, /dev/input/by-path/platform-i8042-serio-0-event-kbd
  Device Number: char 13:67
  Driver Info #0:
    XkbRules: xfree86
    XkbModel: pc104
  Config Status: cfg=new, avail=yes, need=no, active=unknown
`

aidanmcdonagh22, 2021-12-01:
Let me know if you have resolved this :)

hbarcelos, 2021-12-18:
Same issue with A70 MOB:

`sudo hwinfo --keyboard`:

```
36: PS/2 00.0: 10800 Keyboard                                   
  [Created at input.226]
  Unique ID: 9ui9.+49ps10DtUF
  Hardware Class: keyboard
  Model: "AT Translated Set 2 keyboard"
  Vendor: 0x0001 
  Device: 0x0001 "AT Translated Set 2 keyboard"
  Compatible to: int 0x0211 0x0001
  Device File: /dev/input/event3
  Device Files: /dev/input/event3, /dev/input/by-path/platform-i8042-serio-0-event-kbd
  Device Number: char 13:67
  Driver Info #0:
    XkbRules: xfree86
    XkbModel: pc104
  Config Status: cfg=new, avail=yes, need=no, active=unknown
```

Running ```aucc``` I get:

```          
Traceback (most recent call last):
  File "/usr/bin/aucc", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.9/site-packages/aucc/main.py", line 86, in main
    control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
  File "/usr/lib/python3.9/site-packages/aucc/main.py", line 37, in __init__
    super(ControlCenter, self).__init__(vendor_id, product_id)
  File "/usr/lib/python3.9/site-packages/aucc/core/handler.py", line 50, in __init__
    super(DeviceHandler, self).__init__(vendor_id, product_id)
  File "/usr/lib/python3.9/site-packages/aucc/core/handler.py", line 16, in __init__
    self._device = self._get_device(vendor_id, product_id)
  File "/usr/lib/python3.9/site-packages/aucc/core/handler.py", line 27, in _get_device
    if device.is_kernel_driver_active(1):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
```

astiak, 2022-03-03:
Could you please post output of `sudo hwinfo --usb | grep -A 3 8291`?

hbarcelos, 2022-03-03:
> Could you please post output of `sudo hwinfo --usb | grep -A 3 8291`?

For me it's empty.

astiak, 2022-03-03:
@hbarcelos, your keyboard may be incompatible, could you post the output of : 

`lsusb`

and

`sudo dmidecode | grep -i "product\|manufacturer"`

(You may need to install the dmidecode package).

hbarcelos, 2022-03-03:
```
 ~ ➜ lsusb
Bus 004 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. Card Reader
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 024: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 003 Device 005: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 004: ID 04f2:b71a Chicony Electronics Co., Ltd HD Webcam
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
 ~ ➜ sudo dmidecode | grep -i "product\|manufacturer"
	Manufacturer: Avell High Performance
	Product Name: A70 MOB
	Manufacturer: Avell High Performance
	Product Name: Avell A70 MOB
	Manufacturer: Avell High Performance
	Manufacturer: Standard
	Manufacturer: Standard
	Manufacturer: Standard
	Manufacturer: A-DATA Technology
	Module Manufacturer ID: Bank 5, Hex 0xCB
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Manufacturer: A-DATA Technology
	Module Manufacturer ID: Bank 5, Hex 0xCB
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Manufacturer: Intel(R) Corporation
```

astiak, 2022-03-07:
Hi @hbarcelos, doesn't look like your keyboard model is compatible sadly.

I was hoping we could get the model of your chassis and hopefully find another project that may exist. Would you mind posting the output of `lspci`?

hbarcelos, 2022-03-08:
Thanks for taking the time to look into that @astiak.

Here's the output of `lspci`:

```
00:00.0 Host bridge: Intel Corporation 11th Gen Core Processor Host Bridge/DRAM Registers (rev 05)
00:01.0 PCI bridge: Intel Corporation 11th Gen Core Processor PCIe Controller #1 (rev 05)
00:02.0 VGA compatible controller: Intel Corporation TigerLake-H GT1 [UHD Graphics] (rev 01)
00:07.0 PCI bridge: Intel Corporation Tiger Lake-H Thunderbolt 4 PCI Express Root Port #0 (rev 05)
00:08.0 System peripheral: Intel Corporation GNA Scoring Accelerator module (rev 05)
00:0a.0 Signal processing controller: Intel Corporation Tigerlake Telemetry Aggregator Driver (rev 01)
00:0d.0 USB controller: Intel Corporation Tiger Lake-H Thunderbolt 4 USB Controller (rev 05)
00:0d.2 USB controller: Intel Corporation Tiger Lake-H Thunderbolt 4 NHI #0 (rev 05)
00:14.0 USB controller: Intel Corporation Tiger Lake-H USB 3.2 Gen 2x1 xHCI Host Controller (rev 11)
00:14.2 RAM memory: Intel Corporation Tiger Lake-H Shared SRAM (rev 11)
00:14.3 Network controller: Intel Corporation Tiger Lake PCH CNVi WiFi (rev 11)
00:15.0 Serial bus controller: Intel Corporation Tiger Lake-H Serial IO I2C Controller #0 (rev 11)
00:16.0 Communication controller: Intel Corporation Tiger Lake-H Management Engine Interface (rev 11)
00:17.0 SATA controller: Intel Corporation Device 43d3 (rev 11)
00:1c.0 PCI bridge: Intel Corporation Tiger Lake-H PCI Express Root Port #5 (rev 11)
00:1d.0 PCI bridge: Intel Corporation Tiger Lake-H PCI Express Root Port #9 (rev 11)
00:1f.0 ISA bridge: Intel Corporation Tiger Lake-H LPC/eSPI Controller (rev 11)
00:1f.3 Audio device: Intel Corporation Tiger Lake-H HD Audio Controller (rev 11)
00:1f.4 SMBus: Intel Corporation Tiger Lake-H SMBus Controller (rev 11)
00:1f.5 Serial bus controller: Intel Corporation Tiger Lake-H SPI Controller (rev 11)
01:00.0 VGA compatible controller: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
01:00.1 Audio device: NVIDIA Corporation Device 228e (rev a1)
2d:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
2e:00.0 Non-Volatile memory controller: ADATA Technology Co., Ltd. XPG SX8200 Pro PCIe Gen3x4 M.2 2280 Solid State Drive (rev 03)
```

It's an Avell A70MOB.

jariojose, 2022-04-05:
I have the same hardware here. Let me know if you have a solution for this. 

Thank you!

gioufop, 2022-06-01:
I'm having the same problem here:

```
aucc -c green -b 4                                                                                                                                                                                                                  130 ↵
Traceback (most recent call last):
  File "/usr/bin/aucc", line 33, in <module>
    sys.exit(load_entry_point('avell-unofficial-control-center==1.0.4', 'console_scripts', 'aucc')())
  File "/usr/lib/python3.10/site-packages/aucc/main.py", line 86, in main
    control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
  File "/usr/lib/python3.10/site-packages/aucc/main.py", line 37, in __init__
    super(ControlCenter, self).__init__(vendor_id, product_id)
  File "/usr/lib/python3.10/site-packages/aucc/core/handler.py", line 50, in __init__
    super(DeviceHandler, self).__init__(vendor_id, product_id)
  File "/usr/lib/python3.10/site-packages/aucc/core/handler.py", line 16, in __init__
    self._device = self._get_device(vendor_id, product_id)
  File "/usr/lib/python3.10/site-packages/aucc/core/handler.py", line 27, in _get_device
    if device.is_kernel_driver_active(1):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
```

sudo dmidecode | grep -i "product\|manufacturer"
```
	Manufacturer: Avell High Performance
	Product Name: A57
	Manufacturer: Avell High Performance
	Product Name: A57
	Manufacturer: Standard
	Manufacturer: Intel(R) Corporation
	Manufacturer: 0x0080
	Module Manufacturer ID: Bank 1, Hex 0x80
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Manufacturer: 0x0080
	Module Manufacturer ID: Bank 1, Hex 0x80
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
```

sudo hwinfo --keyboard
```
38: PS/2 00.0: 10800 Keyboard
  [Created at input.226]
  Unique ID: c3zD.+49ps10DtUF
  Hardware Class: keyboard
  Model: "AT Translated Set 2 keyboard"
  Vendor: 0x0001 
  Device: 0x0001 "AT Translated Set 2 keyboard"
  Compatible to: int 0x0211 0x0001
  Device File: /dev/input/event4
  Device Files: /dev/input/event4, /dev/input/by-path/platform-i8042-serio-0-event-kbd
  Device Number: char 13:68
  Driver Info #0:
    XkbRules: xfree86
    XkbModel: pc104
  Config Status: cfg=new, avail=yes, need=no, active=unknown
```
uname -a 
`Linux archlinux 5.18.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 30 May 2022 17:53:11 +0000 x86_64 GNU/Linux
`

xevilstar, 2022-06-03:
Same here 

uname -a
Linux ghost 5.18.1-amd64 #1 SMP PREEMPT_DYNAMIC Fri Jun 3 13:01:33 CEST 2022 x86_64 GNU/Linux

    description: Notebook
    product: Standard (0001)
    vendor: PCSpecialist
    version: Standard
    serial: GM7ZN7M2214702145
    width: 64 bits
    capabilities: smbios-3.3.0 dmi-3.3.0 smp vsyscall32
    configuration: boot=normal chassis=notebook family=CEZANNE sku=0001 uuid=6f0cea00-5263-11ec-86ab-ee1d19a62800
  *-core
       description: Motherboard
       product: GM7ZN7M
       vendor: TongFang
       physical id: 0
       version: Standard
       serial: Standard
       slot: Standard
     *-firmware
          description: BIOS
          vendor: American Megatrends International, LLC.
          physical id: 0
          version: N.1.24A10
          date: 08/31/2021
          size: 64KiB
          capacity: 16MiB
          capabilities: pci upgrade shadowing cdboot bootselect socketedrom edd int13floppynec int13floppytoshiba int13floppy360 int13floppy1200 int13floppy720 int13floppy2880 int5printscreen int9keyboard int14serial int17printer int10video acpi usb biosbootspecification uefi
     *-memory
          description: System Memory
          physical id: 13
          slot: System board or motherboard
          size: 64GiB
        *-bank:0
             description: SODIMM DDR4 Synchronous Unbuffered (Unregistered) 2667 MHz (0.4 ns)
             product: CM4X32GE2666C18S2
             vendor: Unknown
             physical id: 0
             serial: 00000000
             slot: DIMM 0
             size: 32GiB
             width: 64 bits
             clock: 2667MHz (0.4ns)
        *-bank:1
             description: SODIMM DDR4 Synchronous Unbuffered (Unregistered) 2667 MHz (0.4 ns)
             product: CM4X32GE2666C18S2
             vendor: Unknown
             physical id: 1
             serial: 00000000
             slot: DIMM 0
             size: 32GiB
             width: 64 bits
             clock: 2667MHz (0.4ns)
     *-cache:0
          description: L1 cache
          physical id: 15
          slot: L1 - Cache
          size: 512KiB
          capacity: 512KiB
          clock: 1GHz (1.0ns)
          capabilities: pipeline-burst internal write-back unified
          configuration: level=1
     *-cache:1
          description: L2 cache
          physical id: 16
          slot: L2 - Cache
          size: 4MiB
          capacity: 4MiB
[... 766 lines cut ...]
       capabilities: i2c
  *-input:6
       product: Power Button
       physical id: 8
       logical name: input3
       logical name: /dev/input/event3
       capabilities: platform
  *-input:7
       product: TUXEDO Keyboard
       physical id: 9
       logical name: input34
       logical name: /dev/input/event22
       capabilities: platform
  *-input:8
       product: Sleep Button
       physical id: a
       logical name: input4
       logical name: /dev/input/event4
       capabilities: platform
  *-input:9
       product: Lid Switch
       physical id: b
       logical name: input5
       logical name: /dev/input/event5
       capabilities: platform

xevilstar, 2023-12-13:
the problem is still present