rodgomesc/avell-unofficial-control-center #64: Eluktronics  Max 17

KJN9000000, 2021-12-06:
This worked great on the Evoo LP6/Eluktronics Mech 17.3 
But the newer thinner chassis called the Max 17 has a different  keyboard it's actually  a much better keyboard  aside from the lights not working with your control  center. They use the same control center in Windows and are manufactured  by the same company  so maybe a driver update to the file would fix the issue.

astiak, 2022-03-03:
Hi, would you mind posting the output of `sudo hwinfo --usb | grep -A 3 8291` please?

KJN9000000, 2022-03-03:
I don't have that one any more I sold it and went with the Mech G3 /
Slimbook Titan. Sorry about that.

Respectfully

Kyle Norris

520-442-7091

***@***.***

https://www.linkedin.com/in/kyle-norris-254b932b

astiak, 2022-03-03:
No worries, thank you for responding.