rodgomesc/avell-unofficial-control-center #70: can't run it

xevilstar, 2022-06-03:
aucc
Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 33, in <module>
    sys.exit(load_entry_point('avell-unofficial-control-center==1.0.4', 'console_scripts', 'aucc')())
  File "/usr/local/lib/python3.10/dist-packages/avell_unofficial_control_center-1.0.4-py3.10.egg/aucc/main.py", line 165, in main
  File "/usr/local/lib/python3.10/dist-packages/avell_unofficial_control_center-1.0.4-py3.10.egg/aucc/main.py", line 104, in __init__
  File "/usr/local/lib/python3.10/dist-packages/avell_unofficial_control_center-1.0.4-py3.10.egg/aucc/core/handler.py", line 50, in __init__
  File "/usr/local/lib/python3.10/dist-packages/avell_unofficial_control_center-1.0.4-py3.10.egg/aucc/core/handler.py", line 16, in __init__
  File "/usr/local/lib/python3.10/dist-packages/avell_unofficial_control_center-1.0.4-py3.10.egg/aucc/core/handler.py", line 27, in _get_device
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'

leonardoprimieri, 2022-08-10:
here too, any updates?

astiak, 2023-04-03:
Hi,

When running `sudo hwinfo --short` do you see the below for keyboard?

```
keyboard:
                       Integrated Technology Express ITE Device(8291)
  /dev/input/event0    AT Translated Set 2 keyboard
```

If so, could you give this a try: https://github.com/astiak/avell-unofficial-control-center/tree/detect-keyboard

xevilstar, 2023-06-13:
I gave it a try .....
which module should I load ?

hwinfo --short|grep ITE
                       Integrated Technology Express ITE Device(8291)
                       Integrated Technology Express ITE Device(8291)

aucc -c green -b 4
Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 33, in <module>
    sys.exit(load_entry_point('avell-unofficial-control-center==1.0.4', 'console_scripts', 'aucc')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/avell_unofficial_control_center-1.0.4-py3.11.egg/aucc/main.py", line 165, in main
  File "/usr/local/lib/python3.11/dist-packages/avell_unofficial_control_center-1.0.4-py3.11.egg/aucc/main.py", line 104, in __init__
  File "/usr/local/lib/python3.11/dist-packages/avell_unofficial_control_center-1.0.4-py3.11.egg/aucc/core/handler.py", line 50, in __init__
  File "/usr/local/lib/python3.11/dist-packages/avell_unofficial_control_center-1.0.4-py3.11.egg/aucc/core/handler.py", line 16, in __init__
  File "/usr/local/lib/python3.11/dist-packages/avell_unofficial_control_center-1.0.4-py3.11.egg/aucc/core/handler.py", line 27, in _get_device
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'