rodgomesc/avell-unofficial-control-center #71: Avell A70 unsupported?

AxeLanderMoreira, 2022-10-04:
sudo hwinfo --short listmy device as:

keyboard:
  /dev/input/event3    AT Translated Set 2 keyboard

When I run `aucc -d`, I get an error:

Traceback (most recent call last):
  File "/usr/local/bin/aucc", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/aucc/main.py", line 86, in main
    control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
  File "/usr/local/lib/python3.8/dist-packages/aucc/main.py", line 37, in __init__
    super(ControlCenter, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python3.8/dist-packages/aucc/core/handler.py", line 50, in __init__
    super(DeviceHandler, self).__init__(vendor_id, product_id)
  File "/usr/local/lib/python3.8/dist-packages/aucc/core/handler.py", line 16, in __init__
    self._device = self._get_device(vendor_id, product_id)
  File "/usr/local/lib/python3.8/dist-packages/aucc/core/handler.py", line 27, in _get_device
    if device.is_kernel_driver_active(1):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'

jariojose, 2023-01-25:
Did you find a solution?

AxeLanderMoreira, 2023-01-28:
Hi @jariojose . Unfortunately not.

At least the Fn+F6 combination works for me, and I can dim down and turn off the lights. But no way to make the control-center actually work.

madebyluque, 2023-09-14:
Also an A70 user, and it is also not working here with the same error @AxeLanderMoreira mentioned.