rodgomesc/avell-unofficial-control-center #72: Is the Avell B.On Lite model supported?

caiopetruccirosa, 2022-12-07:
When running `sudo hwinfo --short` the output is:

```
cpu:                                                            
                       11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz, 3850 MHz
                       11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz, 3769 MHz
                       11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz, 3946 MHz
                       11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz, 3789 MHz
                       11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz, 3992 MHz
                       11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz, 3643 MHz
                       11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz, 3970 MHz
                       11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz, 3900 MHz
keyboard:
  /dev/input/event3    AT Translated Set 2 keyboard
mouse:
  /dev/input/mice      ImPS/2 Logitech Wheel Mouse
  /dev/input/mice      FTCS1000:00 2808:0101 Mouse
  /dev/input/mice      FTCS1000:00 2808:0101 Touchpad
monitor:
                       BOE CQ LCD Monitor
graphics card:
                       Intel UHD Graphics
sound:
                       Intel Tiger Lake-LP Smart Sound Technology Audio Controller
                       C-Media Electronics USB Advanced Audio Device
storage:
                       Intel SATA controller
                       A Non-Volatile memory controller
network:
  wlp0s20f3            Intel Wi-Fi 6 AX201
  enp0s31f6            Intel Ethernet Connection (13) I219-V
network interface:
  wlp0s20f3            Ethernet network interface
  enp0s31f6            Ethernet network interface
  lo                   Loopback network interface
disk:
  /dev/nvme0n1         A Disk
partition:
  /dev/nvme0n1p1       Partition
  /dev/nvme0n1p2       Partition
  /dev/nvme0n1p3       Partition
  /dev/nvme0n1p4       Partition
usb controller:
                       Intel Tiger Lake-LP Thunderbolt USB Controller
                       Intel Tiger Lake-LP USB 3.2 Gen 2x1 xHCI Host Controller
bios:
                       BIOS
bridge:
                       Intel PCI bridge
                       Intel Tiger Lake-LP LPC Controller
                       Intel 11th Gen Core Processor Host Bridge/DRAM Registers
                       Intel 11th Gen Core Processor PCIe Controller
hub:
                       Linux Foundation 2.0 root hub
                       Linux Foundation 3.0 root hub
                       Linux Foundation 2.0 root hub
                       Linux Foundation 3.0 root hub
memory:
                       Main Memory
bluetooth:
                       Intel Bluetooth Device
unknown:
                       FPU
                       DMA controller
                       PIC
                       Keyboard controller
                       PS/2 Controller
                       Intel System peripheral
                       Intel Tiger Lake-LP Serial IO I2C Controller #1
                       O2 Micro SD/MMC Card Reader Controller
                       Intel Signal processing controller
                       Intel Tiger Lake-LP Management Engine Interface
                       Intel Tiger Lake-LP SPI Controller
                       Intel Tiger Lake-LP Serial IO I2C Controller #2
                       Intel Tiger Lake-LP Serial IO I2C Controller #0
                       Intel Tiger Lake-LP Shared SRAM
                       Intel Tiger Lake-LP SMBus Controller
                       Intel Signal processing controller
  /dev/input/event10   Acer BisonCam,NB Pro
  /dev/input/event7    C-Media Electronics USB Advanced Audio Device
```

There is no `Integrated Technology Express ITE Device(8291)` in the keyboard info. But I don't understand what should I do in order to control the keyboard lights, which I imagine that could be controlled since Avell told me that it would work just fine if I used their Control Center.

caiopetruccirosa, 2022-12-07:
I using Pop!_OS 22.04 LTS, which uses Linux kernel version `6.0.6-76060006-generic`.

ToMattBan, 2022-12-14:
Well, I tryied here and it didn't work. When I run it ask for my password and then shows:

`Traceback (most recent call last):
  File "/usr/bin/aucc", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.10/site-packages/aucc/main.py", line 86, in main
    control = ControlCenter(vendor_id=0x048d, product_id=0xce00)
  File "/usr/lib/python3.10/site-packages/aucc/main.py", line 37, in __init__
    super(ControlCenter, self).__init__(vendor_id, product_id)
  File "/usr/lib/python3.10/site-packages/aucc/core/handler.py", line 50, in __init__
    super(DeviceHandler, self).__init__(vendor_id, product_id)
  File "/usr/lib/python3.10/site-packages/aucc/core/handler.py", line 16, in __init__
    self._device = self._get_device(vendor_id, product_id)
  File "/usr/lib/python3.10/site-packages/aucc/core/handler.py", line 27, in _get_device
    if device.is_kernel_driver_active(1):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'`

caiopetruccirosa, 2022-12-26:
@ToMattBan Are you using the Avell B.On Lite model? Are you able to list the device in your laptop?

ToMattBan, 2023-01-01:
@caiopetruccirosa the command `sudo hwinfo --short` shows me
```
keyboard:
  /dev/input/event3    AT Translated Set 2 keyboard
```

It didn't show "Integrated Technology Express ITE Device(8291)"

cristovao-trevisan, 2023-01-06:
Hi. I also have a B.On Lite and was having the same problem. I asked the avell suport center and they told me the keyboard is from Clevo (part number Z6-79-NL50MU0K-001-BE). Found the driver online and it now works for me, you can find the [driver here](https://github.com/tuxedocomputers/tuxedo-keyboard) and [instructions to install it here](https://configurelaptop.eu/clevo-keyboard-backlight-control-for-linux/).

> If you are using Arch (like myself) just install from AUR (`yay tuxedo-keyboard-dkms`) and reboot

ToMattBan, 2023-01-06:
Thank you @cristovao-trevisan 
I needed to install the linux-headers and then the tuxedo package (Using Arch here too) and after a rebbot, it's working. Even the shortcuts works well. 

I just have a question, the colors were to change according to the screen or something like that? I'm incredible happy with it working, it's just a question haha

cristovao-trevisan, 2023-01-06:
The driver repository says there are a few modes, but I didn't work for me. The driver also suports gradient keyboard colors, but that also didn't work. I think our keyboard allows only one color and doesn't support the modes described in the driver repository.
I'm happy enough that I can turn it off now :stuck_out_tongue_closed_eyes: